psiphi75/ahrs #1: Kp and Ki different?

rclai, 2016-08-24:
I noticed [here](https://github.com/psiphi75/ahrs/blob/f015db7684400f5e35a13f6adcc2af0cedc2d814/Mahony.js#L27-L31) that these values are different and on other libraries in other languages the value is different like [here](https://github.com/dccharacter/AHRS/blob/master/MahonyAHRS.c), is there a reason why?

psiphi75, 2016-08-24:
Thanks, it's fixed.  It was due to me testing the Mahony algorithm when I first ported it.  But I tend to use the Madgwick algorithm instead.

closed 2016-08-24 by psiphi75

rclai, 2016-08-24:
Thanks, but on second thought, I just saw that on the [original AHRS code by x-io](https://github.com/xioTechnologies/Open-Source-AHRS-With-x-IMU/blob/5e380626b6d0644d02becde94648b88c48b993e6/x-IMU%20IMU%20and%20AHRS%20Algorithms/x-IMU%20IMU%20and%20AHRS%20Algorithms/AHRS/MahonyAHRS.cs#L78-L85), you can actually pass the `Kp`, `Ki`, or both optionally and they default to `1.0` and `0`, respectively. Is that something you can build in?

psiphi75, 2016-08-24:
Fixed.  There were other problems with the algorithm which have been fixed as well.  Default value for `Kp`/`Ki` are `1.0`/`0.0`.

rclai, 2016-08-24:
Cool, thanks.

commit 440d8426 (2016-08-24): Fixed #1, changed default values