psiphi75/ahrs #3: Possible variable wrong?

rclai, 2016-08-25:
[These three variables](https://github.com/xioTechnologies/Open-Source-AHRS-With-x-IMU/blob/5e380626b6d0644d02becde94648b88c48b993e6/x-IMU%20IMU%20and%20AHRS%20Algorithms/x-IMU%20IMU%20and%20AHRS%20Algorithms/AHRS/MahonyAHRS.cs#L276-L278) and [these three variables](https://github.com/psiphi75/ahrs/blob/47f63b5e633de2c400ecd19f19d0750c555954fb/Mahony.js#L109-L111) are different.

rclai, 2016-08-25:
I have confirmed that yes, those values need to be:

```js
qa = q1;
qb = q2;
qc = q3;
````

Instead of

```js
```js
qa = q0;
qb = q1;
qc = q2;
````
```

psiphi75, 2016-08-25:
The JavaScript code uses variables q0 ... q3, where as the original C code uses q1 ... q4.  But the algorithm should be the same.  I don't see the issue here.  If you do find the algorithm is different, let me know.

rclai, 2016-08-25:
I think you're right. Interestingly, though the code was not ported identically, it still yields the same result.

closed 2016-08-25 by rclai

psiphi75, 2016-08-25:
Thanks.  I first started porting the original code, it worked, but then I based it on the code [here](https://github.com/ZiCog/madgwick.js), although the code was rather messy, so I created my own repo.  But the ZiCog repo was good for testing purposes.