psiphi75/ahrs #4: Update Issues

KennyBarraud, 2016-08-26:
Hello.
I'm having some trouble with this library when I update the values of the madgwick AHRS filter.
My application is a web app and i'm using Cordova to get it into Android. To get the acceleration I use that [plugin](https://cordova.apache.org/docs/en/latest/reference/cordova-plugin-device-motion/index.html) which gives me a vector 3 in m/s² and that [plugin](https://www.npmjs.com/package/cordova-plugin-gyroscope) to get the gyroscope. But when I get the euler angles after having the filter updated, the values are set to NaN.

So here's my question, what are the units of what you need to put into your update function ?
Am I using this library correctly ?

psiphi75, 2016-08-26:
That's a good point, because units are not documented.  I have updated the documentation.
The units are:
- gyroscope: radians / s
- accelerometer: g, where 1 g is 9.81 m/s²
- magnetometer: unit-less, but a relative proportion of the Earth's magnetic field

Do you provide magnetometer data as well?  If you don't then you can't supply the `deltaTimeSec` value. 

It would be good to see the code you are using to initialise and call the plug.

KennyBarraud, 2016-08-29:
Thank you for your answer !
Here's the code I have done :
```
    onmove: function(){
        if(this.isActive === true) {
            navigator.accelerometer.getCurrentAcceleration(this.onSuccessAcceleration.bind(this), this.onError.bind(this));
        }
    },

    onSuccessAcceleration: function (acceleration) {
        this.acceleration = acceleration;
        this.gyroscope.getCurrent(this.onSuccessDirection.bind(this), this.onError.bind(this));
    },

    onSuccessDirection: function(directions){
        this.directions = directions;
        this.madgwick.update(this.directions.x, this.directions.y, this.directions.z, this.acceleration.x, this.acceleration.y, this.acceleration.z);
        navigator.geolocation.getCurrentPosition(this.moveCamera.bind(this), this.onError.bind(this), this.application.geolocationOptions);
    },

    moveCamera: function(position){
        var position = ADV.GEO.projection.convertToPositionDegree(position.coords.latitude, position.coords.longitude, 0);
        this.application.mainCamera.position.set(position[0], position[1], position[2]);
        var orientation = this.madgwick.getEulerAngles();
        this.application.mainCamera.rotation.set(orientation.z, orientation.x, orientation.y);
    }
```

The onmove function is called for every render frame (about 125 ms).
I also give you an exemple about the results of the differents functions which get the accelerometer and the gyroscope :

this.directions :
x: -0.25652483105659485
y: -0.2624049782752991
z: 3.314098834991455

this.acclerations:
x: 7.2142720222473145
y: 3.555708169937134
z: 7.643636226654053

To correct that, I tried to divide the result of the accelerometer by 9.81 to get it in g and not in m/s². But I still got a NaN result. Am I doing it right ?

psiphi75, 2016-08-29:
Closed with this fix: https://github.com/psiphi75/ahrs/commit/d5101113215749648f6dbbe2a768f1dd80de20da
Issue was due to blatant copying from C code to JavaScript.  JavaScript allows for undefined function parameters, these were not checked for and caused the issue.

closed 2016-08-29 by psiphi75

pliablepixels, 2017-10-29:
@KennyBarraud I just noticed this closed issue. I am also using cordova, but a different plugin (it helps normalize the data b/w iOS and Android) but the concept is the same.

Questions for you:
1. Are you using x/y/z of accelerometer without the effect of gravity?
2. What are you supplying as the alpha/beta/gamma values? Absolute angles in radians or change in rotation? (device orientation value vs device motion value). thx