psiphi75/ahrs #5: Unexplainable drifting in output

JohnMcLear, 2017-05-23:
My X degrees value is drifting despite the sensor being completely static..

degrees line is output from:

```
var data = new Int8Array(buffer);

    var accX = (new Int16Array([data[0] | data[1] << 8]))[0];
    var accY = (new Int16Array([data[2] | data[3] << 8]))[0];
    var accZ = (new Int16Array([data[4] | data[5] << 8]))[0];

    var gyrX = (new Int16Array([data[6] | data[7] << 8]))[0];
    var gyrY = (new Int16Array([data[8] | data[9] << 8]))[0];
    var gyrZ = (new Int16Array([data[10] | data[11] << 8]))[0];

    accX = (accX / 32768)*8;
    accY = (accY / 32768)*8;
    accZ = (accZ / 32768)*8;

    gyrX = (gyrX / 32768)*500;
    gyrY = (gyrY / 32768)*500;
    gyrZ = (gyrZ / 32768)*500;

    madgwick.update(accX, accY, accZ, gyrX, gyrY, gyrZ);

    var radian = madgwick.toVector();

    var x = radian.x * 180 / Math.PI;
    var y = radian.y * 180 / Math.PI;
    var z = radian.z * 180 / Math.PI;

    console.log("degrees", x, y, z)
```

and the second line is the input accX, accY, accZ, gyrX, gyrY, gyrZ...

```
degrees 55.25852317577883 -10.09981475143741 -11.282539981329702
-0.0185546875,-0.012451171875,-0.009033203125,0.1983642578125,1.40380859375,0.396728515625
degrees 55.23403493658435 -10.494481711302507 -11.039637142907287
-0.02197265625,-0.009765625,-0.010498046875,0.0762939453125,1.40380859375,0.42724609375
degrees 55.198606987402464 -10.790160863101603 -10.93126548988808
-0.0185546875,-0.00830078125,-0.01025390625,0.1068115234375,1.434326171875,0.4730224609375
degrees 55.20341090457497 -10.411313854885528 -11.26917558330974
-0.021484375,-0.0107421875,-0.0107421875,-0.091552734375,1.4495849609375,0.48828125
degrees 55.16524767990943 -10.640732351469085 -11.241735339812616
-0.020263671875,-0.012939453125,-0.016357421875,0.1220703125,1.40380859375,0.518798828125
degrees 55.13786071972521 -10.634825687622843 -11.380823689777705
-0.018310546875,-0.010009765625,-0.0126953125,0.06103515625,1.373291015625,0.4730224609375
degrees 55.145719728034216 -10.232311673781636 -11.707080912825134
-0.019775390625,-0.01123046875,-0.010986328125,-0.06103515625,1.373291015625,0.457763671875
degrees 55.14794035180655 -9.835888561961204 -12.031887680742933
-0.0205078125,-0.010986328125,-0.00732421875,-0.0762939453125,1.3885498046875,0.4730224609375
degrees 55.14231271082064 -9.44210230755254 -12.368443836130773
-0.020263671875,-0.010986328125,-0.010498046875,-0.030517578125,1.28173828125,0.4425048828125
degrees 55.15047973540168 -9.213314137024902 -12.503830516257265
-0.02197265625,-0.012451171875,-0.011962890625,-0.0457763671875,1.3580322265625,0.396728515625
degrees 55.145733209366874 -8.80251702552961 -12.81679183057671
-0.016845703125,-0.00830078125,-0.012451171875,-0.152587890625,1.3885498046875,0.457763671875
degrees 55.15717184116389 -8.746371136747301 -12.806003921377565
-0.019287109375,-0.00927734375,-0.00927734375,-0.0152587890625,1.3885498046875,0.4425048828125
degrees 55.15689569595431 -8.42572431449599 -13.020383134665217
-0.024169921875,-0.011474609375,-0.01171875,-0.0457763671875,1.40380859375,0.4425048828125
degrees 55.14456565501209 -8.016884640488167 -13.32714483274315
-0.01708984375,-0.0078125,-0.013916015625,-0.1373291015625,1.3885498046875,0.4425048828125
degrees 55.154984946527534 -8.233678739026061 -13.150685152590926
-0.021240234375,-0.011962890625,-0.010498046875,0.030517578125,1.4190673828125,0.3662109375
degrees 55.11791895918746 -8.297798468732177 -13.265289299238605
-0.016357421875,-0.015380859375,-0.012939453125,-0.030517578125,1.373291015625,0.457763671875
degrees 55.10334738417022 -7.892785262571427 -13.569502491622282
[... 1970 lines cut ...]
degrees 41.946015279340685 -22.81147068013321 -31.66977987935002
-0.02001953125,-0.00927734375,-0.013427734375,-0.1678466796875,1.40380859375,0.4730224609375
degrees 41.947433988739945 -22.45741718668554 -31.91995527996887
-0.019775390625,-0.01123046875,-0.012939453125,-0.18310546875,1.46484375,0.54931640625
degrees 42.046715074284094 -22.247032846880835 -31.93665027806642
-0.01806640625,-0.0107421875,-0.0048828125,-0.06103515625,1.5411376953125,0.48828125
degrees 41.89481913576826 -22.428015944972742 -32.009601377790666
-0.01904296875,-0.010009765625,-0.012939453125,0.0457763671875,1.4190673828125,0.5340576171875
degrees 41.97132337411186 -22.48085791273858 -31.87201580924453
-0.019287109375,-0.01025390625,-0.00927734375,-0.0152587890625,1.4801025390625,0.4425048828125
degrees 42.06202254446799 -22.19578296980451 -31.952149033167522
-0.01904296875,-0.009765625,-0.010986328125,-0.1678466796875,1.617431640625,0.4730224609375
degrees 42.11231709542163 -21.84776351091351 -32.12529110276817
-0.019287109375,-0.0126953125,-0.00927734375,-0.4425048828125,1.7852783203125,0.5645751953125
degrees 42.11515901946381 -21.49115920980741 -32.361239262360286
-0.019287109375,-0.013916015625,-0.01220703125,-0.1678466796875,1.5716552734375,0.54931640625
degrees 41.96092019902721 -21.652768568179887 -32.45373845022258
-0.024658203125,-0.010498046875,-0.01171875,0.1068115234375,1.52587890625,0.67138671875
degrees 41.9552451932304 -21.92635414599569 -32.276907296890144
-0.020751953125,-0.013671875,-0.02197265625,0.213623046875,1.4495849609375,0.2899169921875
degrees 42.05961940995741 -21.67375504543724 -32.311655905132184
-0.02001953125,-0.004638671875,-0.006103515625,-0.3204345703125,1.3275146484375,0.274658203125
```

Note the drift of 13 degrees...  Any idea what I'm doing wrong?

psiphi75, 2017-05-23:
It looks like there is no magnetometer (compass).  This will stop the angular drift.  The reason is that the gyroscope data is very noisy and the values will eventually drift.

Using a magnetometer is the the only way to stop drift entirely.  However you can reduce drift by:
 - Calibrating the gyroscope bias (biggest source of drift) and accelerometer, see [here](https://github.com/miniben-90/mpu9250#calibration) and [the calibration code](https://github.com/miniben-90/mpu9250/blob/master/calibrate_gyro.js) and [how it's applied](https://github.com/miniben-90/mpu9250/blob/master/mpu9250.js#L367).  With a well calibrated device it will drift, but much slower.  Note that the gyroscope is also temperature sensitive, so it's bias changes with temperature.
- Play with the `beta` value for the *Madgwick* algorithm, or the `kp` and `ki` values for the *Mahony* algorithm.  However, by reducing drift you increase latency, or reaction time.

JohnMcLear, 2017-06-08:
After quite a bit of time staring at this code I discovered the drift is not the root cause of the problem.  I'm getting pretty much unusable / unreliable values irrespective which means I think I'm piping in incorrect numbers..  I did basic Phi/Theta computation and get semi-usable numbers but I want to use AHRS to get more reliable numbers..  This is my output of a few IMU readings / computation:

```
{ ax: 0.9912109375,
  ay: 0.1201171875,
  az: 0.1787109375,
  gx: 0.396728515625,
  gy: -0.762939453125,
  gz: -0.152587890625,
  phi: -6.80093969961194,
  theta: -77.7437039898772,
  rawData: { ax: 4061, ay: 492, az: 732, gx: 27, gy: -49, gz: -9 } }
{ angle: 1.9131248874302929,
  x: -0.9782544019855904,
  y: -0.15202728846053132,
  z: -0.1410887258399942 }
{ ax: 0.9921875,
  ay: 0.12158203125,
  az: 0.17578125,
  gx: 0.3662109375,
  gy: -0.91552734375,
  gz: -0.1220703125,
  phi: -6.880070515150009,
  theta: -77.8434152024247,
  rawData: { ax: 4064, ay: 498, az: 721, gx: 25, gy: -60, gz: -8 } }
{ angle: 1.9019002320604812,
  x: -0.9743777948966866,
  y: -0.18268641897718887,
  z: -0.13120055309929682 }
```

Note that the angles coming from AHRS are nothing like Phi/Theta..

I'm doing:

```
  madgwick.update(data.gx, data.gy, data.gz, data.ax, data.ay, data.az);
  console.log(madgwick.toVector());
```

So it makes sense to not get similar values with that so I switch up to

```
  madgwick.update(data.gx, data.gy, data.gz, data.ax, data.ay, data.az);
  console.log(madgwick.getEulerAngles());
```

And I get:
```
{ ax: 0.99169921875,
  ay: 0.12060546875,
  az: 0.1787109375,
  gx: 0.48828125,
  gy: -0.885009765625,
  gz: -0.1220703125,
  phi: -6.825099730316998,
  theta: -77.73450777570605,
  rawData: { ax: 4062, ay: 494, az: 733, gx: 32, gy: -57, gz: -8 } }
{ heading: -3.0117845123070373,
  pitch: -1.17129345147732,
  roll: -1.9609205412222732 }
{ ax: 0.990234375,
  ay: 0.12158203125,
  az: 0.1787109375,
  gx: 0.579833984375,
  gy: -0.6103515625,
  gz: -0.152587890625,
  phi: -6.889597923916833,
  theta: -77.68665587049568,
  rawData: { ax: 4056, ay: 499, az: 733, gx: 38, gy: -39, gz: -10 } }
{ heading: -2.9398821638464008,
  pitch: -1.1969071414667525,
  roll: -1.9943619929444278 }
{ ax: 0.99072265625,
  ay: 0.12255859375,
  az: 0.177734375,
  gx: 0.457763671875,
  gy: -0.67138671875,
  gz: -0.18310546875,
  phi: -6.94229616363923,
  theta: -77.70655383548201,
  rawData: { ax: 4059, ay: 503, az: 729, gx: 30, gy: -43, gz: -12 } }
{ heading: -2.8524759463954314,
  pitch: -1.2210703134796024,
  roll: -2.0496362594401334 }
```

**Note that the heading heading/pitch/roll and roll barely go above 2 degrees**..

And with this in mind I feel I'm piping AHRS the wrong values..  I'm passing it the values in the object that **isn't** the rawData object..  

Any thoughts?

psiphi75, 2017-06-08:
I have to say I don't know what Theta and Phi do.  But it looks like your gyro numbers are very large.  The algorithm is expecting a value in radians per second, are your values in degrees per second?  It's also really important to adjust for bias on the gyroscope, see my previous comment on this.  Bias on the gyro is a huge contributor to rotational drift.

JohnMcLear, 2017-06-09:
Interestingly we got Roll and Pitch stable but Heading is wildly erratic and difficult to understand/debug.  For now for my use case I might be able to use Roll/Pitch without Heading so I might be able to survive..  

FWIW the fix was as you proposed..

```
  data.gx = Math.radians(data.gx);
  data.gy = Math.radians(data.gy);
  data.gz = Math.radians(data.gz);
```

psiphi75, 2017-06-10:
What do you consider "wildly erratic".  It's expected the heading will drift after a while without a magnetometer in place.

Thanks for the donation offer, but I don't have anything set up.

JohnMcLear, 2017-06-17:
By wildly erratic I mean it does a 180 degree drift within 87 seconds.  Does this sound normal?
[image]

Pretty picture that illustrates event included :)

psiphi75, 2017-06-17:
Have you compensated for bias in the gyroscope?  This will reduce your error, see my [previous comment](https://github.com/psiphi75/ahrs/issues/5#issuecomment-303532218).  However, without a magnetometer (compass) drift is normal, it's always going to occur.  You should also be using the Magwick filter, it works better without a compass that the Mahony filter.

Have a look at the [test code here](https://github.com/psiphi75/ahrs/blob/master/test/test.js#L26), set lines 26 and 27 both to true.  Then run `node test` from the command line.  Two tests will be run, both use the same data, run for around 6 seconds and use Magwick.  The tests have real sample data from an IMU, the MPU9250.  The first test has the magnetometer enabled, the second test has it disabled.  You will notice that the test without the compass drifts by an extra 3.5 degrees over the 6 seconds.

JohnMcLear, 2017-06-19:
oh okay so 0.6 degree drift p/s is normal.  That's super interesting, I for some reason assumed drift would be in the order of degrees per minute not dps so that's good to know.  I will get on with these changes but probably not in client side software but attempt to make the compensation in firmware.

psiphi75, 2017-06-19:
Actually, with good calibration around bias, you should be able to get it
to degrees per minute.  I have to say that i can't remember if those
examples values were compensated for bias or not.

JohnMcLear, 2017-08-12:
We solved this by doing this

``
 sampleInterval: 1000 / sampleIntervalFromFile,
``

TLDR; our sampleInterval was completely wrong...  We were passing 104..

closed 2017-08-12 by JohnMcLear

psiphi75, 2017-08-14:
Great stuff.  Thanks for the update.