psiphi75/ahrs #6: Fast initialization

jessetane, 2017-08-04:
Thanks for this module, it works very well once my magnetometer input has stabilized. I noticed the TODO in the readme, I'm curious if you already had an approach in mind?

psiphi75, 2017-08-04:
I haven't delved into it deeply.  But, the problem is that quaternion coordinate framework starts at one particular orientation, at (1, 0, 0, 0), [here is the init](https://github.com/psiphi75/ahrs/blob/a95ab7d5ab3adaf433143a4408109973fa42d221/Madgwick.js#L32).  For the first iteration it should initialise these values by converting the first Cartesian coordinate received to a quaternion.  This is the ideal approach, but I don't have time to figure out the Math since it's not an issue for me.

Another approach (a.k.a ugly hack) is to temporally relax the the `beta` or `kp`/`ki` values.  Yet, another approach (a.k.a ugly hack) is on the first iteration is to do a small loop to re-calculate the numerous times, this is something that the consuming application can do quite easily.

jessetane, 2017-08-04:
As amusing as it would be to dig into the math I unfortunately don't have time right now either, the loop approach works well enough for now, thanks!

jessetane, 2017-08-04:
switched to animating to my desired beta from an initially high value. stabilizes in less than 1 second which is good enough for this particular app. had to alter the module a bit to expose the property which _might_ be worth pushing back upstream even though it's not the real solution.

Lelelo1, 2020-04-19:
Animating beta helped me too. I thought there was too bad sensor quality that did it. For some reason it took around 20 seconds to get accuracy in some orientations. It's always around 0.7 second now!

first-dev, 2021-09-15:
Hey @Lelelo1 , how did you animate beta, can you please show a quick example.
I tried using setTimeout like so:
```
private ahrsConfig = {
  algorithm: 'Madgwick',
  sampleInterval: this.sampleInterval,
  beta: 10,
}

private ahrs = new AHRS(this.ahrsConfig)

setTimeout(() => {
  this.ahrsConfig = {
    algorithm: 'Madgwick',
    sampleInterval: this.sampleInterval,
    beta: 0.1,
  }
  console.log(`updated beta`)
}, 2000)
```
but the angles stay unstable

psiphi75, 2021-09-15:
You can't animate `beta` without updating the code.

You could add the following `setBeta()` function to Madgwick.js:
```
  return {
    update: madgwickAHRSUpdate,
    setBeta(b) { beta = b; },
    getQuaternion() {
      return {
        w: q0,
        x: q1,
        y: q2,
        z: q3,
      };
    },
```

By the way, I'm looking at updating the code to make initialisation trivial.

psiphi75, 2021-09-23:
This is is now resolved.  The brute-force initialisation is no longer.  The quaternion is correctly initialised when `doInitialisation` is set to `true`.

commit e966a5b1 (2021-09-23): Correctly initialise the internal quaternion - closes #6
closed 2021-09-23 by psiphi75