psiphi75/ahrs #12: Axis rotation

kenobi91, 2019-06-19:
Hello dear friend,
i was playing around with this module since I wanted to get quaternion then rotation matrix and then acceleration in world frame and I keep getting strange results. I then came back and saw that my axis are wrong, if I understood you correctly this is axis layout:

[image]

And i want to use this to calculate from data on smartphone, my smartphone has these axis:

[image]

I am not sure if these axis has some sort of marking like XYZ or YXZ and I am having difficulties to understand how can axis from smartphone can be rotated into axis for this algorithm. I would really appreciate if you can just point me where to look for. Thanks :)

psiphi75, 2019-06-20:
I've updated the documentation to give a clearer idea of the orientation.  

For the phone to match the orientation of the AHRS algorithm, you can rotate every sensor's values around the z-axis, 90 degrees anti-clockwise.  This can be done like this:

```JavaScript
const rotatedAccel = {
    x: accel.y,
    y: -accel.x,
    z: accel.z
}
```

kenobi91, 2019-06-20:
Thank you very much, I saw yesterday, one issue where you guys discuss this.

closed 2019-06-20 by kenobi91