psiphi75/ahrs #13: a question about the package

MuhammadLebda, 2019-06-30:
sorry to bother you with such a stupid question. 
is this package reads the sensors values ( accelerometer, gyroscope) then calculates the orientation? or just takes those values as an input and then calculates the orientation ? because when I tried to run the code, I get an error " gyro is not defined " 
thank you :)

psiphi75, 2019-07-01:
This package will not read the values, you need to find a package that does.  Then pass those values to this package.

MuhammadLebda, 2019-07-01:
aha, thank you :) 
do you happen to know if there's any good package to read the sensors' values ?

psiphi75, 2019-07-02:
That depends on your device, every device will have its own API interface. I suggest you find the JavaScript API documentation for your device and go from there.

closed 2021-08-23 by psiphi75