psiphi75/ahrs #14: Using an offset to overcome hinder with pitch in Euler angles?

Lelelo1, 2019-12-30:
In vertical - as if a photo is taken on an iPhone in portrait - pitch is near -90 degrees. And 0 deg when laying on a table. This is when using the setup:
```
const rotatedAccel = {
    x: accel.y,
    y: -accel.x,
    z: accel.z
}
```

In my current app I am developing where I am displaying AR objects onto to a camera view. These drastically change in position badly when pitch extends the vertical position (as if looking upwards with the mobile camera).
Heading is flipped -so that a north heading with [0, 360] suddenly becomes 180 degrees. 
Roll is affected the same.

Due to the fact pitch is [90, -90] I can't detect wether the user is looking upwards, and adjust the AR objects.
`-77` deg could either be looking upwards or downwards.

So I thought I might be throwing this out here in case anyone might be stumbling into a similar issue.
Maybe there is someway of turning axises so that instead of [0, 0, 0] being `---` flat on table could be `/`?

Lelelo1, 2020-01-01:
I got around the issue by starting up a second `Madgwick`, that had the following setup: 

```
const rotatedAccel = {
    x: accel.x,
    y: accel.y,
    z: accel.z
}
```

There I could use i'ts pitch, to use as my roll instead. So looking up, or looking down - rolling produces the same, consistent value.

In my setup I still use the original Roll to have heading stay in front of the user irregardless of orientation: [landscape-left], [portrait], [landscape-right], so if north, any phone orientation results heading of 0 by doing `let adjustedHeading = Heading - Roll`.

As the new Roll provided good results on i'ts own, I never really came in need of pitch. But a pitch that goes [180, -180] so that up/down can be handled better can be gotten by the `Roll` value of the second Madgwick instance.

So it can be that the [suggested coordinate system](url) is not practical to work with.

closed 2020-01-01 by Lelelo1

psiphi75, 2020-01-06:
Thanks for raising the issue and resolving it.  You are right that the default orientation will not fit all use cases out-of-the-box.  This is because the of the [Gimble Lock](https://en.wikipedia.org/wiki/Gimbal_lock) problem.  One way around the gimble lock is to use quaternions using the `getQuaternion()` function provided.  Most good frameworks will using quaternion orientation internally and provide an API exposing this.