psiphi75/ahrs #16: Calculate sample interval

bknill, 2020-10-18:
Forgive my igorance. 

I have a sample rate of  4.93 millis from my GoPro. 

How do I covert that to Hz? 

Thanks

psiphi75, 2020-10-18:
```
Frequency in Hertz (Hz) = 1 / Sample Rate (seconds)
                   Freq = 1 / (Sample Rate (milliseconds) / 1000)
                   Freq = 1 / (4.93 / 1000)
                   Freq = 202.8 Hz
```

bknill, 2020-10-18:
Thanks.  Weirdly it works well when sampleRate = 4.93 and does crazy stuff when it's at 202.8

I'm using this with extracted data from Go Pro telemetry. i.e. looping through the samples and updating AHRS every sample. 

```
  const sampleInterval = 202.8
    const converttoG = ms2 => ms2/9.80665
    const times = Object.keys(samples)
    const madgwick = new AHRS({sampleInterval, beta: 0.5, doInitialisation: false})
```
```
      const data = Object.keys(samples).reduce((prev, key, index) =>  {
            const {gyro, accel, magn} = samples[key]
        
            if(magn){
                init.magn = magn
                return prev
            }
            //return {...prev, [key]: [gyro[0], gyro[1],gyro[2]]}
            init.gyro = gyro || init.gyro
            init.accel = accel || init.accel
            //Data order Z,X,Y
            madgwick.update(
                init.gyro[1],init.gyro[2], init.gyro[0], 
                converttoG(init.accel[1]), converttoG(init.accel[2]), converttoG(init.accel[0])
                )
            const {heading, pitch, roll} = madgwick.getEulerAngles()
            // heading is from north, going west (about z-axis).
            // pitch is from vertical, going forward (about y-axis).
            // roll is from vertical, going right (about x-axis).
            return {...prev, [key]: [heading, pitch, roll]}
        }, {} )  
```

I'm then using this data in Three.js to rotate a sphere with the footage projected onto for a VR stabalisation effect. 

When the sample rate is 4.93 this works well but is a bit jerky. 

Am I going about this the right way? 

Another question, the magnetmeter data comes back with 4 values, any idea how to convert this into a compass direction!?

bknill, 2020-10-18:
ps. Hi from Ben in Wellington!

psiphi75, 2020-10-18:
Hi Ben from Wellington.

I don't know much about Go Pro telemetry.  Yes, the `sampleInterval` in this case should be 4.93, although 5.0 will suffice.  You may want to use the following:

```JavaScript
madgwick.update(
                init.gyro[1],init.gyro[2], init.gyro[0], 
                converttoG(init.accel[1]), converttoG(init.accel[2]), converttoG(init.accel[0]),
                undefined, undefined, undefined, 
                deltaTimeSec
                )
```
Where `deltaTimeSec` is the time since the last update.  This can help when data is not arriving exactly at `sampleInterval`, or you skip data points.  

Regarding the magnetometer, what does the documentation say it returns?  Does it give you the units?

bknill, 2020-10-18:
Thanks! 

I'm using https://github.com/JuanIrache/gopro-telemetry which returns the units in µT

A sample looks like this [-550, -71, -41, 6991]

psiphi75, 2020-10-18:
Have a look to find more information about what each value in the array is.

bknill, 2020-10-18:
Looks like a problem on the MAX https://github.com/gopro/gpmf-parser/issues/82

bknill, 2020-10-19:
The values are three axis x,y,z and rhall in microteslas

psiphi75, 2020-10-19:
You can just plug those magnetometer x, y and z values straight into `madgwick.update`.

closed 2021-08-23 by psiphi75