psiphi75/ahrs #17: GetEuler angle always gives values in range of -3 to 3. Am i missing something?

nazneenhuma, 2021-03-14:
i see that you have defined your axis with x pointing north and z up, my accelerometer has z down as shown below
[image]

You say to rotate the outputs in such case, how do i go about doing that? i mean i did not understand what rotate the outputs mean

psiphi75, 2021-03-14:
> GetEuler angle always gives values in range of -3 to 3. Am i missing something?

The output is in radians.  Multiple each value by `180 / 3.14159` to get degrees.

> You say to rotate the outputs in such case, how do i go about doing that?

In your case you need to [rotate the axes](https://en.wikipedia.org/wiki/Rotation_of_axes) by 180 degrees around X or Y.  Rotating by 90/180 degrees is pretty simple math. This will probably work for you:

`update(-g.x, g.y, -g.z, -a.x, a.y, -a.z, -m.x, m.y, -m.z)`.  This will rotate the axis around the y-axis by 180 degrees, so Z will point up (opposite to before) and X will also point in the opposite direction.

Please let me know if this solves your problem.

nazneenhuma, 2021-03-24:
I tried the above and thot may be magnetometer may also be reason so currently running with 6 axis without passing in magnetometer values but however i see the value decreasing when object is stable suppose say it starts with 300, next value is 298 ,then 290, then 270 then 264 etc.. 
Not sure what would be the problem

nazneenhuma, 2021-03-24:
This is how i get values with device stabke.. no movement at all
angle is 9.221062
 angle is 12.700016
 angle is 14.068754
 angle is 14.036126
 angle is 13.220181
 angle is 12.837092
 angle is 12.737936
 angle is 12.333611
 angle is 12.340543
 angle is 12.092478
 angle is 11.896714
 angle is 11.957815
 angle is 11.596505
 angle is 11.866184
 angle is 11.187619
 angle is 10.918409
 angle is 10.837001
 angle is 10.722542
 angle is 10.457922
 angle is 10.304446
 angle is 10.153737
 angle is 10.000969
 angle is 9.617128
 angle is 9.294669
 angle is 9.297976
 angle is 8.946549
 angle is 9.011792
 angle is 8.617909
 angle is 8.673428
 angle is 8.167043
 angle is 8.239808
 angle is 7.797703
 angle is 8.034293
 angle is 7.384024
 angle is 7.632738
 angle is 7.109461
 angle is 7.360253
 angle is 6.692821
 angle is 7.000992
 angle is 6.334758
 angle is 6.678481
 angle is 5.862691
 angle is 6.302578
 angle is 5.582349
 angle is 6.003461
 angle is 5.317854
 angle is 5.630772
 angle is 4.793670
 angle is 5.317567
 angle is 4.337935
 angle is 4.936696
 angle is 3.880060
 angle is 4.558188
 angle is 3.482072
 angle is 4.243866
 angle is 3.000194
 angle is 3.821285
 angle is 2.660140
 angle is 3.494411
 angle is 2.268433
 angle is 3.269067
 angle is 1.978467
 angle is 3.025944
 angle is 1.704951
 angle is 2.759019
 angle is 1.321890
 angle is 2.376791
 angle is 0.957729
 angle is 2.077674
[... 494 lines cut ...]
 angle is 274.383392
 angle is 276.177979
 angle is 274.035309
 angle is 275.853455
 angle is 273.701355
 angle is 275.493744
 angle is 273.391663
 angle is 275.118713
 angle is 273.040894
 angle is 274.798187
 angle is 272.824799
 angle is 274.542542
 angle is 272.573669
 angle is 274.229065
 angle is 272.253052
 angle is 273.970886
 angle is 271.947906
 angle is 273.529022
 angle is 271.577911
 angle is 273.292297
 angle is 271.265747
 angle is 272.924957
 angle is 271.029572
 angle is 272.605957
 angle is 270.568420

psiphi75, 2021-03-24:
Yes, without a magnetometer the values will drift slowly. The reason is that the gyroscope is noisy, reducing this noise by calibration will improve things, but won't remove your problem. The magnetometer provides a fixed reference to a real world orientation and the AHRS algorithm can use this to overcome drift you are experiencing.

nazneenhuma, 2021-03-24:
True that.
But will it drift so much? Within 1 minute I see values going from 0 to 360 twice..

psiphi75, 2021-03-24:
If it’s not calibrated well, or at all, then yes.  Also, I can imagine some
gyros are noisier than others, do you get the same result for multiple
units?

nazneenhuma, 2021-03-24:
I have calibrated the gyro and the accelerometer. 
And yes, haven't tried with other units. Just tried on a single unit. 
Let me try on others and get back to you. 
I actually thot its something wrong that i might have done in the code. I am using LSM9DS1 currently.

psiphi75, 2021-03-24:
If you attach the raw output for 2 minutes, from the sensors I can have a look. You need to include the time each set of values was captured.

psiphi75, 2021-08-23:
Am closing due to no reply.

closed 2021-08-23 by psiphi75