psiphi75/ahrs #18: Is this possible to get the position difference or the distance the device was moved from the initial point?

arthedza, 2021-08-23:
Or at least the device rotation angle from the initial attitude.

psiphi75, 2021-08-23:
> Is this possible to get the position difference or the distance the device was moved from the initial point?

No, not with this library. I have to say I don’t know what library is available for that. Doing it by first principals should work.

> Or at least the device rotation angle from the initial attitude.

Same as previous answer. Using first principles you would sum the gyroscope values for each axis and multiple by elapsed time.

arthedza, 2021-08-23:
Thank you for your fast response. And I have one more question: what algorithm I should choose for this purpose?

psiphi75, 2021-08-23:
That depends what you want to do.  What angle of rotation do you want to measure and about which axis?  Can you make the angle of rotation always be around one axis only?  Then you can use this library.  You just start measuring the angles.

closed 2021-08-25 by arthedza