psiphi75/ahrs #20: Get user heading

dmenneck, 2022-05-31:
Hi @psiphi75, thanks for your great package! It looks very promising.

I'm working on a smartphone-based pedestrian dead reckoning system and for that i need the heading direction of the user. The literature often states that a Madgwick and Mahony filter could help with obtaining the user's heading. 

Sadly i'm not quite sure how to interpret the values obtained from `madgwick.getEulerAngles()`.

```js
madgwick.update(gyr.x, gyr.y, gyr.z, acc.x, acc.y, acc.z, mag.x, mag.y, mag.z);
const { heading } = madgwick.getEulerAngles()

console.log(heading * (180/Math.PI))
```

I'd like to draw conclusions like 'the user turned 45 degrees to the left'. Do you think it's possible with this package?

psiphi75, 2022-06-01:
> Do you think it's possible with this package?

That depends.  If you use this on a mobile phone and it's in the user's pocket, then it's unlikely to help.  But if you place the IMU on a stable platform that is always positioned the same way relative to the direction of motion, it will be useful. 

Essentially you need to know the orientation of the IMU with respect to the frame of reference for the object you are measuring.  See here https://github.com/psiphi75/ahrs#orientation.  I.e. if the user is always holding the mobile device in the same orientation then the heading will work if you have the correct orientation.

> Sadly i'm not quite sure how to interpret the values obtained from `madgwick.getEulerAngles()`.

Again, you need to convert the orientation of your [IMU values](https://github.com/psiphi75/ahrs#orientation) (e.g. ax, ay, yz, gx, ...) to the orientation of the body for `madgwick.getEulerAngles()` to make sense.

psiphi75, 2022-06-15:
Sorry for the delay in the reply, I thought I had answered this.  

> So It’s very likely that the smartphone will move. Would you say that it’s stable enough, @psiphi75?

That's something you would need to test yourself.  "Stable enough" is pretty subjective and depends a lot on your application.   I would say you should be able to get +/-15 degrees.  If you use any compass app on your phone, you would get a feel for it.

psiphi75, 2022-10-11:
> The smartphone will be in a hand hold position, in front of the body of
> the user. So It’s very likely that the smartphone will move. So it’s
> probably not possible to get accurate heading estimations?
>
> The initial orientation will be known, because the user has to position
> himself in the direction of the pathway.

It depends how accurate you want to be.  I would say you should be able to
get +/-15 degrees.  If you use any compass app on your phone, you would get
a feel for it.