psiphi75/ahrs #24: Sorry, I have a low-level problem. I don't understand how to use it in a browser. I tried to use it this way, but the result was incorrect

msdog, 2024-02-20:
```js
import AHRS from "ahrs";
import { Vector3 } from "three";
const rad2deg = 180.0 / Math.PI;
const deg2rad = Math.PI / 180.0;

const madgwick = new AHRS({
  /*
   * The sample interval, in Hz.
   *
   * Default: 20
   */
  sampleInterval: 20,

  /*
   * Choose from the `Madgwick` or `Mahony` filter.
   *
   * Default: 'Madgwick'
   */
  algorithm: 'Madgwick',

  /*
   * The filter noise value, smaller values have
   * smoother estimates, but have higher latency.
   * This only works for the `Madgwick` filter.
   *
   * Default: 0.4
   */
  beta: 0.4,

  /*
   * The filter noise values for the `Mahony` filter.
   */
  kp: 0.5, // Default: 0.5
  ki: 0, // Default: 0.0

  /*
   * When the AHRS algorithm runs for the first time and this value is
   * set to true, then initialisation is done.
   *
   * Default: false
   */
  doInitialisation: false,
});
let gyro = new Vector3()
let acceleration = new Vector3()
let compass = new Vector3()

window.addEventListener('deviceorientation', (event) => {
  const alpha = event.webkitCompassHeading != null ? event.webkitCompassHeading : 360 - event.alpha;
  compass.x = alpha
  compass.y = event.beta
  compass.z = event.gamma
})
window.addEventListener('devicemotion', (event) => {
  gyro.x = event.rotationRate.alpha * deg2rad
  gyro.y = event.rotationRate.beta * deg2rad
  gyro.z = event.rotationRate.gamma * deg2rad

  acceleration.x = event.acceleration.x
  acceleration.y = event.acceleration.y
  acceleration.z = event.acceleration.z
  update()
})

function update() {
  madgwick.update(
    gyro.x,
    gyro.y,
    gyro.z,
    acceleration.x,
    acceleration.y,
    acceleration.z,
    compass.x,
    compass.y,
    compass.z,
  );

  console.log(madgwick.getEulerAnglesDegrees());
}

msdog, 2024-02-20:
I am developing an indoor map navigation web application and would like to use this library to correct the problem of inaccurate compass interference caused by magnetic fields

psiphi75, 2024-02-20:
What error messages are you getting? ‘import’ likely doesn’t work, that’s
new JS syntax. You may just need to use ‘require’ instead, search for “amd
js require”.

msdog, 2024-02-25:
> 您收到什么错误消息？“导入”可能不起作用，那是 新的 JS 语法。您可能只需要使用“require”来代替，搜索“amd js require”。

I am using the Rollup package plugin to use 'import' normally, but the compass head data obtained from using this plugin is incorrect

psiphi75, 2024-02-25:
I need more information about this.  Error messages, example data, what the actual problem is, etc.

closed 2025-07-03 by psiphi75